Juerd/ESP-WiFiSettings #7: SSID with apostrophe is truncated

brentward, 2020-07-13:
Using version 3.3.0 of the library on an ESP-WROOM-32 board, specifically the AWS ESP32 Lanyard, the SSID is being truncated at an apostrophe character.

Steps to reproduce:
code:
`#include <SPIFFS.h>
#include <WiFiSettings.h>

void setup() {
  Serial.begin(115200);
  SPIFFS.begin(true);  // On first run, will format after failing to mount

  WiFiSettings.connect();

}

void loop() {
  Serial.println("Hello World");
  delay(2000);
}`

1. Start up device.
2. Connect to configuration network
3. Select the desired SSID: `House Bettis's Wi-Fi Network`
4. Enter the proper password
5. Click Save
6. Get an error that `House Bettis` is not in range

Juerd, 2020-07-13:
Thanks for reporting this!

I see where it went wrong: line 57 includes `'` in the range of safe-to-use characters, but it should of course be escaped if I'm putting html attributes in single quotes!

I'll update it right away!

Juerd, 2020-07-13:
Note: I haven't actually tested the fix but am confident that it should work now :)

commit 6955be55 (2020-07-13): Exclude quotes from list of safe characters in html_entities().

Fixes #7
closed 2020-07-13 by Juerd