Juerd/ESP-WiFiSettings #9: arduino-esp32: Local IP is 255.255.255.255 after connection succeeds

santosrafael, 2020-10-19:
Hi,

After connecting to AP, I get the message "Connecting to WiFi SSID 'acesspoint'..255.255.255.255", so something went wrong and no connections could be established.

When I commented the [line 346](https://github.com/Juerd/ESP-WiFiSettings/blob/6955be55c0e71f536d8d15445700f6b832752377/WiFiSettings.cpp#L346) in file WiFiSettings.cpp the issue was gone.

Juerd, 2020-10-19:
Is that on an ESP32 or ESP8266? Are you using the latest version of the ESP platform for Arduino; if not, which version are you using?

Juerd, 2020-10-29:
Sorry, I've not been able to reproduce this problem. Can you share the smallest program you can make, that still shows the problem?

Juerd, 2021-06-14:
No reply after >½ year; closing.

closed 2021-06-14 by Juerd