Juerd/ESP-WiFiSettings #16: Feature Request: Manual SSID

OptimusGREEN, 2021-04-04:
Hi,

It would be great to be able to manually enter the ssid, this would allow hidden ssid's to be used.

Thanks.

Juerd, 2021-04-04:
This is (by definition :P) a highly obscure feature, and would complicate the user interface to an extent that it is likely to confuse regular users or require some tricks to hide it from sight (javascript?) until explicitly requested. I'm not enthusiastic about the suggestion, and think that if someone is techie enough to use hidden SSIDs, they can probably just write to the SPIFFS file manually too.

OptimusGREEN, 2021-04-04:
I was thinking it would be a case of just adding "custom" to the ssid dropdown, which when selected allowed text input in the box.

My ssid's for my not and iot are just hidden so the mrs etc don't try to connect to it lol. 

If I could code arduino i'd try myself but it hurts my head :)

Juerd, 2021-04-04:
> I was thinking it would be a case of just adding "custom" to the ssid dropdown, which when selected allowed text input in the box.

It's a web page. A `<select>` element doesn't allow for manual input, so that would require an `<input type=text>` element. However, for reasons outlined in my comment above that would either complicate the user interface with an extra element or require javascript to hide it until necessary. That adds complexity and code which in the case of small platform isn't free. If this were a very common thing, I'd probably build support for it, but it is extremely uncommon and almost every use of hidden WiFi networks is misguided anyway, so I'd rather spend my time educating people who think they're useful that in fact they're not.

> My ssid's for my not and iot are just hidden so the mrs etc don't try to connect to it lol.

I'm sure they can handle the cognitive load of having to click the right network. Most people in the world have to do so, and it's not generally considered much of a barrier. Where I live, there are over 50 networks in range, of which only 2 are mine.

There is no harm in trying to connect to the wrong network, as long as the attempt fails. I suggest using a password that they don't have, in order to have the attempt fail. Alternatively, just tell them which one is the right one, or give the networks useful names like `this one` and `not this one` and `not this one either`.

> If I could code arduino i'd try myself but it hurts my head :)

If neither of us is going to implement this feature, or has any idea on how to do it in a way that doesn't overly complicate the software, I think it makes sense to just close it. Thanks for taking the time to post your suggestion, though. I do appreciate it.

closed 2021-04-04 by Juerd