Juerd/ESP-WiFiSettings #21: Compilation error with platform espressif8266 version 3.1.0

notsleep196, 2021-07-16:
Hello!
After updating espressif8266 platform to the latest version 3.1.0 can't compile my sketch. I have error, located in ESP-WiFiSettings\WiFiSettings.cpp file. 
The error output:

> .pio\libdeps\d1_mini_E22-400M30S\ESP-WiFiSettings\WiFiSettings.cpp: In member function 'void WiFiSettingsClass::begin()'
:
.pio\libdeps\d1_mini_E22-400M30S\ESP-WiFiSettings\WiFiSettings.cpp:11:35: error: expected ')' before '__INT32'
   11 |     #define ESPMAC (Sprintf("%06" PRIx32, ESP.getChipId()))
      |                                   ^~~~~~

Environment:
    - wemos d1 mini
    - CLion + platformio
    - platform espressif8266 version 3.1.0

notsleep196, 2021-08-02:
Hello, @Juerd!
Any ideas about this issue?

Juerd, 2021-08-07:
Sorry, I am currently not available. In case you find a fix, please submit a pull request.

Juerd, 2021-08-07:
Coincidentally I came across a mention of a bug in the ESP8266 Arduino Core introduced in version 3.0.0 and fixed in 3.0.2, that's probably the cause of this issue. Please upgrade and test again. If that doesn't fix it, feel free to reopen this issue.

https://github.com/esp8266/Arduino/pull/8222

closed 2021-08-07 by Juerd

notsleep196, 2021-08-09:
Hello, @Juerd!
I tried to use development version of espressif8266 platform with Arduino 3.0.2 in it. The error is still the same. 
Please re-open the issue, i don't have enough access rigths