Juerd/ESP-WiFiSettings #22: Getting 255.255.255.255 as IP address

MarcEngrie, 2021-08-10:
Hi,

I came across your nice library and started testing it. Getting the portal and doing what is need is fine however when WiFi network is configured I saw 2 things

First after succesfull config followed by a reboot, connection to my WiFi network failed. Doublechecked pw and that's not the issue. 
19:28:53.054 -> Connecting to WiFi SSID Engrie.............................. failed.
19:29:10.139 -> Starting access point for configuration portal
But, this one might not be related to your lib as I have it with the normal WiFi library as well where I need a second reboot to get connected. 
However at the second reboot gives me an IP (I guess) of 255.255.255.255. 

19:29:18.989 -> Connecting to WiFi SSID Engrie.255.255.255.255

I came across that one some weeks ago after upgrading to the new core. The solution was simple but re-create a previous issue. See my comments in one of my apps written 

  // Next line was needed before Core 1.0.5 to set the hostname correctly
  // since Core 1.0.5 rc6 no longer. see https://github.com/espressif/arduino-esp32/issues/4732
  // leaving the line will result in getting 255.255.255.255 as IP-address.
  // WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE, INADDR_NONE);
  // removing line above curres getting ip 255.255.255.255 but
  // brings back the issue of hostname not being set in DHCP -  DAMMED!!!!
  WiFi.mode(WIFI_STA);
  WiFi.setHostname(hostname);
  WiFi.begin(wifiSSID, wifiPassword, WIFI_ALL_CHANNEL_SCAN);
  
Is this related or do I have another problem?

didier9, 2021-11-13:
I have the same issue here with an M5Stamp-PICO (ESP32-PICO-D4) running the Basic example.
I got the setup page the first time and entered my settings and rebooted.
The device does connect to the access point but does not get an IP address and the serial monitor reports 255.255.255.255.
I have looked at the library and tried to force some static settings (ip, gateway, netmask), replacing the call WiFi.config( INADDR_NONE, INADDR_NONE, INADDR_NONE ); but that did not work at all.
My Arduino expertise is quite limited, so I stopped there.
Interestingly, now that the settings are in Flash, the Advanced example does connect and does get an IP address.
Also as a bonus, the router properly identifies the device as esp32-C30B00

pitersonfascist, 2022-01-08:
I have the same issue with TTGO T-Display board.
Surprisingly, when I compile the same sketch on MacOS with the same library version it works

academo, 2022-02-20:
I was also having this problem with nodemcu esp32 which is also described here https://github.com/Juerd/ESP-WiFiSettings/issues/9

Commenting this line fixed it for me https://github.com/Juerd/ESP-WiFiSettings/blob/master/WiFiSettings.cpp#L469

Which is a workaround mentioned here https://github.com/espressif/arduino-esp32/issues/2537#issuecomment-508558849 but it is not longer working because it was fixed here https://github.com/espressif/arduino-esp32/commit/ad4cf1461bea3fb1378402b572383813ea9c2a06

This means the library needs some logic to detect which version of the platform is running to apply the workaround or not. 

As @Juerd indicated in other issues he is not longer working on the library (understandable) so I downloaded it and patched it for my project which makes it work well.

Juerd, 2024-10-28:
Going by the last comment, this issue seems to have been fixed in the esp32 arduino project. Thanks to the reporter for reporting the issue, and thanks to commenters for sharing updates!

closed 2024-10-28 by Juerd