Juerd/ESP-WiFiSettings #24: Didn't work with ESP32 C3 version

Skywalkerf34, 2021-12-11:
Hi,

I just try advanced and OTA version form your sample.
I did compile and run
Serial give me AP and SSID ESP-xxxxx and a password
But this device is not really on air ????

DTTerastar, 2021-12-14:
I just got a C3 to try it on myself.

Sys64736, 2021-12-16:
Worked good on my esp32 C3.   Using devkitc-02   

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381cc4
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1430
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d68
SHA-256 comparison failed:
Calculated: a42fd66d3efcfe06041f22310384e34152b441ee2a556807ecf4bb04bd2bf039
Expected: 4c5ab38f8781ab61391d70bf9c7e3acc52c2fa2977802cef9ae40657a00dda4b
Attempting to boot anyway...
entry 0x403ce000
I (53) boot: ESP-IDF v4.4-dev-3235-g3e370c4296 2nd stage bootloader
I (53) boot: compile time 12:14:25
I (54) boot: chip revision: 3
I (54) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (60) boot.esp32c3: SPI Speed      : 80MHz
I (64) boot.esp32c3: SPI Mode       : DIO
I (68) boot.esp32c3: SPI Flash Size : 4MB
I (71) boot: Enabling RNG early entropy source...
I (76) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (85) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (91) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (98) boot:  2 app0             OTA app          00 10 00010000 00300000
I (104) boot:  3 spiffs           Unknown data     01 82 00310000 000f0000
I (111) boot: End of partition table
I (114) boot_comm: chip revision: 3, min. application chip revision: 0
I (120) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=16958h ( 92504) map
I (143) esp_image: segment 1: paddr=00026980 vaddr=3fc8e200 size=032a8h ( 12968) load
I (145) esp_image: segment 2: paddr=00029c30 vaddr=40380000 size=063e8h ( 25576) load
I (151) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=9e24ch (647756) map
I (255) esp_image: segment 4: paddr=000ce274 vaddr=403863e8 size=07c60h ( 31840) load
I (261) esp_image: segment 5: paddr=000d5edc vaddr=50000000 size=00010h (    16) load
I (266) boot: Loaded app from partition at offset 0x10000
I (266) boot: Disabling RNG early entropy source...
Connecting to WiFi SSID NOTYOURS....................................................................................................................................................................................................................................................................................................... failed.
Starting access point for configuration portal.
SSID: 'esp32-2049af'
192.168.4.1
[ 51158][E][WebServer.cpp:635] _handleRequest(): request handler not found
[ 52657][E][WebServer.cpp:635] _handleRequest(): request handler not found
[ 52800][E][WebServer.cpp:635] _handleRequest(): request handler not found
21 WiFi networks found.
[ 58999][E][WebServer.cpp:635] _handleRequest(): request handler not found
21 WiFi networks found.
[ 70825][E][WebServer.cpp:635] _handleRequest(): request handler not found
19 WiFi networks found.

I purposely put in the wrong pass for the chosen SSID for that test.

Skywalkerf34, 2021-12-16:
@Sys64736 

Compliation was working fine for me too, the thing is I didn't find the Access point I setUp on air with sniffer.....did you have this "on air" ? then I will check again.
As I am using original example, something maybe wrong in my environment /setup then

Sys64736, 2021-12-16:
No, it's not.  I thought it was connected, as it states in the serial but I was actually seeing my other C3 listed in the router.   Debug states an IP of 255.255.255.255 so somethings off.
I'm going to look at it a little closer tomorrow and see if I can figure it out.

Sys64736, 2021-12-16:
@Skywalkerf34 
It's working for me now and getting a valid IP.  

Comment out the line below in WiFiSettings.cpp
**Line:469**     WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);  // arduino-esp32 #2537

That line disables the DHCP on the router from handing out an IP.   I think it was an issue from a while ago in regards to hostnames and router leases with multiple ESP32's connected.   Hoping it's fixed now with the latest IDF

Skywalkerf34, 2021-12-16:
OK, so This was exactly what I get myself....everything seem running on debug (serial) but nothing "on air"

Skywalkerf34, 2021-12-16:
@Sys64736 
Thanks Marc,
I will try your patch soon and let you know what results on my side

Sys64736, 2021-12-16:
Ok Francis, sounds good.

Skywalkerf34, 2021-12-20:
@Sys64736

I comment the line 469 as you suggest then
I try both basic and OTA example with same result :
compile good with no warning, serial give SSID but still cannot find this "on air"

Sys64736, 2021-12-21:
I don't know why you can't see the access point.  Did you try another phone or computer with a WiFi radio.   It does show up "on air" but sometimes you have to wait for a network list to update from a scan... usually 20-30 seconds but depends on the device.

Skywalkerf34, 2022-01-15:
@Sys64736 
Happy new year !
I try with an ESP32WROOM with basic example and it works great!
Witch mean that this is ESPC3 issue is is because single core ???

mitra42, 2024-10-28:
The code works fine on my ESP Lolin C3 Pico, are you still having problems.

Juerd, 2024-10-28:
Going by the last few comments, this issue seems to have been fixed in the esp32 arduino project. Thanks to the reporter for reporting the issue, and thanks to commenters for sharing updates!

closed 2024-10-28 by Juerd