Juerd/ESP-WiFiSettings #25: ESP32 does not connect unless restarted through portal

frogcrush, 2021-12-17:
On multiple ESP32's on multiple different Wi-Fi networks, I cannot get the device to connect to Wifi unless I hit "restart" through the portal interface. If I remove the timeout, it never connects and just fills the serial output with periods.

NetForces, 2022-04-26:
@frogcrush you can do this:

```
    WiFiSettings.onConfigSaved  = []() {
      ESP.restart(); // Reboot as soon as config is saved
    };
```

Will reboot as soon as config is saved.

Just started using this library and I like it very much. Good job @Juerd

Juerd, 2023-04-10:
@frogcrush I believe the issue you ran into may have been a problem in the esp32 arduino framework. If this is still relevant, however, please reopen the issue.

closed 2023-04-10 by Juerd