Juerd/ESP-WiFiSettings #26: Drop down list on custom variables

JoaquimBoavida, 2022-04-15:
Hi,
This is nor a real issue but a request for help.
The library works fine, it is simple and as a good looking interface. I use custom variables with success.
However I need a variable that must be on drop down list (use must choose from one of 20 or more values) on the portal.
This will be similar to the one that appears when we choose the SSID.
It is any way to implement it? I have tried with HTML but without success.
I don' have the skills to modify the library..

Thanks in advance for any help.

Joaquim

Juerd, 2022-04-15:
This is currently not supported. The drop down for SSIDs is implemented with different code than the stuff used for the custom variables.

I also have no plans to support this.

If you want this but can't provide an implementation yourself, you could use a numeric field (`.integer()`) with the text feature (like `.info()`) to provide the available options.

JoaquimBoavida, 2022-04-16:
Thanks for you fast reply. I will do that.
One other question. The "Restart device" button does not really works in my case. I have to resent (or power cycle) to effectively restart. When i press restart, the integrated LED flashes for some milliseconds but the device does not restart. I'm using WEMOS LOLIN D1 mini (ESP8266).
Thanks again for you help and your library.

DTTerastar, 2022-04-16:
My fork can do this

Juerd, 2022-04-16:
@DTTerastar You have an interesting approach to open source. You fork a project, then don't contribute anything back, but you do use the original project to advertise the fork.

I like some changes you made and will probably reintegrate them. However, it would have been nice if you had contacted me beforehand because I'll do some things differently and there will now be a split in the road forever.

The license absolutely allows what you're doing but I'm not very happy with it.

DTTerastar, 2022-04-17:
> @DTTerastar You have an interesting approach to open source. You fork a project, then don't contribute anything back, but you do use the original project to advertise the fork.
> 
> I like some changes you made and will probably reintegrate them. However, it would have been nice if you had contacted me beforehand because I'll do some things differently and there will now be a split in the road forever.
> 
> The license absolutely allows what you're doing but I'm not very happy with it. Forking is great for an experimental branch or when the original project is dead or doesn't want any of your changes, but you seem to be taking forking as the FIRST option, and that's just annoying.

I actually did attempt to contribute back and you didn't want it.  My response here was to inform you that I had added it.  If you would like me to contribute it back, I am happy to do so.  But I don't want to waste my time if you don't want it.

Juerd, 2022-04-17:
@DTTerastar My apologies, you did indeed contribute PR #23; I neglected to check the closed PRs. As you can tell there are several things I liked and some that needed work. I don't know why you chose to close that PR without opening a new one as you had suggested in your comment there.

In any case, I'll close this issue for now as the question has been answered.

closed 2022-04-17 by Juerd

DTTerastar, 2022-04-17:
I have definitely taken my fork down a road that I don't think you want.  BUT I can contribute the dropdown code if you'd like.  Would you like that?

JoaquimBoavida, 2022-04-17:
Guys,

 

I am sorry for all the commotion I caused. I’m an average programmer and I don’t have the skill for advanced work, like you guys.

 

In the end I was confused.. Darrel toke the fork from the repository and Juerd dind’t  update is form Darrel’s work.. right?

 

In the end, even if there is a solution nobody can help?

 

C’mon guys.. I will buy you both a beer… 😊

 

Joaquim

Enviada: 17 de abril de 2022 03:24
Para: Juerd/ESP-WiFiSettings ***@***.***>
Cc: JoaquimBoavida ***@***.***>; Author ***@***.***>