Juerd/ESP-WiFiSettings #28: Wifi AP hostname

bobnecat, 2022-09-28:
I know this library is pretty old and most likely no longer maintained but I find it to be the most versatile library among all the WiFi connection managers for ESP32. 

The issue that I have is that I cannot set the AP name of ESP32 device using the **WiFiSettings.hostname** function. Maybe I am doing something wrong but would really appreciate if someone could guide me with an example on how to properly add hostname (AP name) definition  function in my arduino sketch. Thank you!

Juerd, 2022-09-28:
bobnecat skribis 2022-09-28 15:53 (-0700):
>I know this library is pretty old and most likely no longer maintained but I find it to be the most versatile library among all the WiFi connection managers for ESP32.

I do intend to maintain it. It's just not broken enough yet, and for my own needs, it's feature complete.

>The issue that I have is that I cannot set the AP name of ESP32 device using the **WiFiSettings.hostname** function. Maybe I am doing something wrong but would really appreciate if someone could guide me with an example (...)

WiFiSettings.hostname = "example";

Or, if you want the device id appended:

WiFiSettings.hostname = "example-";

This assignment should be done before the first WiFiSettings function call.

bobnecat, 2022-09-28:
Thanks for the quick reply! I suppose I did it wrong and instead of declaring hostname I tried passing it as a function. 

Was doing _WiFiSettings.hostname("example-")_ instead of _WiFiSettings.hostname = "example-";_ 

I found your library to be the most reliable out of 3-4 different libraries I've used in the past! Great work and thank you for support!

On a side note, is there a way to time-out the captive portal and cause ESP to restart after some time? I know you have a connect timeout but would be great to have a captive portal timeout as well. The reason for this is that in case if the power goes out it takes for ESP32 to load a lot faster than the router so the library does not see active wifi point that it knows (even with connect timeouts if router takes too long to load) and once it enters the captive portal it just never leaves. Would be great to have it automatically restart the ESP after 5-10 minutes of "inactivity" thus avoiding forever in captive portal state. maybe you already have such a function but I cannot make it out from your guide.

Juerd, 2022-09-28:
bobnecat skribis 2022-09-28 16:13 (-0700):
>I found your library to be the most reliable out of 3-4 different libraries
>I've used in the past! Great work and thank you for support!

Thanks you and you're welcome!

>On a side note, is there a way to time-out the captive portal and cause ESP to restart after some time?

Sure, just put something in the onPortalWaitLoop:

WiFiSettings.onPortalWaitLoop = []() {
     if (millis() > 30000) ESP.restart();
}

> Would be great to have it automatically restart the ESP after 5-10 minutes of "inactivity" thus avoiding forever in captive portal state.

The example above would restart after 30 seconds of uptime, unconditionally. Of course, you'd probably want to hook into onPortalView and onConfigSaved and keep a variable that tracks the last activity for your use case. You'll have to make that yourself.

Note that any user activity that doesn't result in an HTTP request to the ESP wouldn't be taken into account if you do this. The page would still be open, and you wouldn't notice until you tried to do something like save the settings. You could add some custom javascript to fetch the portal page in the background when there's activity in the browser...

WiFiSettings.html("script", "javascriptHere();", false);

Have fun :)

bobnecat, 2022-09-28:
Perfect! thanks again!

closed 2022-09-28 by bobnecat