Juerd/ESP-WiFiSettings #29: Cannot connect to soft AP after .portal() is called

ameerhazo, 2022-12-07:
Hello, I'd like to be able to reconfigure some parameters on the portal even after initial configuration by having the portal brought up again after a button press. The button press function to open the portal works but for some reason, I'm not able to open the portal after successfully connecting to the soft AP.  The .portal() function was called in the loop function when the button is pressed for 3 seconds.  Thanks in advance for the help! 

Regards, 
Ameer

ameerhazo, 2023-03-22:
Hi @Juerd, could you please help me with my problem here :D Would be greatly appreciated.

ameerhazo, 2023-03-22:
Hello @Juerd, thanks for the fast reply. In the API reference section, you wrote,

> void portal();

> Normally, this function is called by .connect(). To allow reconfiguration after the initial configuration, you could call .portal() manually, for example when a button is pressed during startup.

I wish it would work like this. Unfortunately, when this function is called, the portal AP is opened and I'm able to connect to the AP but accessing the AP on the browser gives me a HTTP 500 error.

Juerd, 2023-03-22:
> I wish it would work like this. Unfortunately, when this function is called,
> the portal AP is opened and I'm able to connect to the AP but accessing the AP
> on the browser gives me a HTTP 500 error.

It works for me, e.g. in https://github.com/revspace/operame/blob/main/operame.ino#L182

I don't know why it doesn't work in your case, and I can't debug remotely.

ameerhazo, 2023-03-22:
Hi @Juerd thanks for the code example, I'll try it out on my end to see if it works for me. Thank you and have a nice evening :D

closed 2023-04-10 by Juerd