Juerd/ESP-WiFiSettings #30: Compile error with expansion of macro 'F'

Richhoef, 2023-05-07:
Whenever I try to compile this library, I get compile errors saying it cannot convert `const char []` to `const __FlashStringHelper*` in assignment `#define F(string_literal) (string_literal)`.

This is my Platformio.ini file:
```
[env:esp32thing]
platform = espressif32
board = esp32thing
framework = arduino
lib_deps =
    juerd/ESP-WiFiSettings@^3.8.0
```

And this is the file being compiled:
https://github.com/Juerd/ESP-WiFiSettings/blob/master/examples/Basic/Basic.ino

The errors:
[image]

Can anyone help me?

Juerd, 2023-05-07:
This is due to a recent breaking change in arduino-esp32.

A fix is available at https://github.com/espressif/arduino-esp32/pull/8147. You can apply that manually, wait for the next release of arduino-esp32, or downgrade arduino-esp32 to the last working version.

Richhoef, 2023-05-07:
Ah, that explains. Thanks!

qguv, 2023-05-09:
You can jump the queue by telling platformio to use a particular commit of the platform package, see [this commit](https://github.com/revspace/operame/commit/e550ae4a67093edc86073433a4ed7b9187bf8c4c) for an example.

qguv, 2023-05-09:
@Richhoef: You can jump the queue by telling platformio to use a particular commit of the platform package, see [this commit](https://github.com/revspace/operame/commit/e550ae4a67093edc86073433a4ed7b9187bf8c4c) for an example.

Richhoef, 2023-05-15:
> @Richhoef: You can jump the queue by telling platformio to use a particular commit of the platform package, see [this commit](https://github.com/qguv/operame/commit/e550ae4a67093edc86073433a4ed7b9187bf8c4c) for an example.

Thanks!

Juerd, 2024-10-28:
Going by the last few comments, this issue seems to have been fixed in the esp32 arduino project. Thanks to the reporter for reporting the issue, and thanks to commenters for sharing updates!

closed 2024-10-28 by Juerd