Juerd/ESP-WiFiSettings #35: WDT conflict with ESP32 support v3.x.x

dquadros, 2024-06-12:
**Not sure if this is a problem with the library or the ESP32 support from Espressif.**

When trying the minimal usage example with the esp32 package v 3.0.1 and an ESP32-C3 board, I get continuously the error

`task_wdt: esp_task_wdt_reset(763): task not found`

Removing the call to esp_task_wdt_reset() in WiFiSettings.cpp fixes the problem.

Trying the minimal usage example with an ESP32 DEVKIT V1 board, I always get the error, with or without the call to esp_task_wdt_reset().

epgodoy, 2024-08-06:
The same problem here with Heltec ESP32 LoRa board. Any help?

mitra42, 2024-10-28:
I posted a duplicate issue #36 which I'll close .... 
Doing a bit of research on this at . https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/wdts.html
It looks like the task is not subscribed to the ESP watchdog function. so there is no need to reset it and the reset is generating an error as a result - harmless, just makes the Serial noisy and hard to debug.
I would suggest either removing the line, or - if its needed - subscribing the task to the timer.

I'm not going to do a PR because there is quite likely a reason that line is included without a subscription. I'm also not fully sure I understand the functionality or the different ways this might be getting used.

Juerd, 2024-10-28:
The watchdog reset was added because it was needed, all the way back in the first version.

It is unclear to me why it now breaks with some boards, and why it doesn't always break. It seems that the problem is ESP32 specific, but I don't currently have all the different boards available to test things, and that makes me reluctant to change anything without fully understanding the problem.

Specifically, I don't know if removing that line will maybe fix it for some boards, while breaking it for other boards.

mitra42, 2024-10-28:
My (uneducated) guess, is that ts NOT a board problem, but usage, e.g. in some projects the library is being used as part of a ESP task (e.g. using ESP-IDF), and because it sits in a loop the idle task watchdog complains, and the esp_task_wdt_reset keeps it happy, while in other projects (like mine) its standalone through the Arduino-IDE ?

dquadros, 2024-10-28:
> My (uneducated) guess, is that ts NOT a board problem, but usage, e.g. in some projects the library is being used as part of a ESP task (e.g. using ESP-IDF), and because it sits in a loop the idle [...]

My (uneducated) guess is that this a consequence of the changes Espressif made in ESP32 Arduino v3.xx. Maybe they stopped subscribing the Arduino main loop to the ESP Task watchdog.

I have not found (yet) time to investigate further but I do have boards with the various ESP32 models (the original one, S2, S3, C3 and C6). At the time I open the issue, I got the library working on a ESP32-C3 board by removing the line, but got the error message with a ESP32 board regardless of the presence of the line (suggesting that someplace else is calling esp_task_wdt_reset when the CPU is an ESP32).

Anyway, simply removing the line will not fix the problem with version 3.xx and will probably break the library for previous Arduino ESP32 versions.

I will update this issue if I find more information.

dquadros, 2024-10-28:
A made a few tests and could not reproduce the problem with ESP32 Arduino v 3.01 and a original ESP32 with the line commented out.

mitra42, 2024-10-29:
So I wonder if the ESP32 then reboots if the line is commented out (because its sitting in an endless loop, and the wdt triggers?)

dquadros, 2024-11-07:
Maybe this is related to this issue: https://github.com/espressif/arduino-esp32/issues/10547

Notice that they are saying that "the idle task WDT was disabled it on the second core". I've asked for more information on that.