Juerd/ESP-WiFiSettings #37: Unable to use phone to connect node to its hotspot

mitra42, 2024-10-26:
Thanks for the library - super useful and saved a bunch of time for us. 

I'd like to test devices with a hotspot. 

But. .... if the device comes up with the portal, and I connect to it from the phone, 
Then the phone's hotspot isn't shown (because the phone turns it off when connecting to the node's). 

I've then tried accessing the node form my laptop - that's problematic - I couldn't get it to work at all in Firefox (requires https) or Safari (redirected to www.192.168.4.1 !), but it worked in Chrome. But this showed another problem - that the rescan process is faster than the default broadcast frequency of my iPhone, so I had to hit "Rescan" multiple times before it randomly saw the phone. 

After that the device worked fine - but its too hit or miss to use in the field. 

I don't have any great solutions - just flagging it as a problem. 
(p.s. the project we are using it on is [FrugalIoT](https://github.com/mitra42/frugal-iot) and the code is in [system_wifi.cpp](https://github.com/mitra42/frugal-iot) where we have a watchdog on the portal, so it occasionally retries to connect to a known WIFi. )

Juerd, 2024-10-28:
Using a phone's hotspot is an interesting chicken-and-egg problem if you don't have a second phone to configure the ESP. The way Firefox and Safari apparently behave isn't helping either.

If you're configuring devices in bulk, it may be easier to pre-fill the configuration files in the SPI filesystem using esptool.py.

Juerd, 2024-10-28:
The only solution I can think of, is to allow manual entry of the SSID. But that leads to a can of worms regarding my preference for a KISS user interface... I have to think about this.

mitra42, 2024-10-28:
Yes -that was what I was thinking - it could be a config option to add a blank (like on my laptop, I can choose "Other" and enter a SSID/PW that doesn't show up in the SSID scan) 

Also ... probably worth making the scan take longer, another problem i hit was that when trying to connect to the phone's SSID I had to hit Rescan a random number of times till it shows up, I'm guessing that phones (or maybe just iPhones) don't broadcast their SSID as often. (I hadn't yet posted an issue as was planning to experiment first)