Juerd/ESP-WiFiSettings #40: HTML server

mitra42, 2024-11-01:
The core server should be: 

- [ ] Based on the node/express pattern used for mitra.biz. 
- [ ] Be able to connect to a mosquitto server, either local or remote 
- [ ] Able to subscribe to data and store in a CSV or similar based on some (Yaml) configuration
- [ ] Able to serve the HTML client
- [ ] Thinks thru privacy and security issues

Use case / test scenario..
- [ ] User connects to server, opens a index for a specific org, picks project from drop-down, 
- [ ] Has the active sensors popular 
- [ ] Interacts with the actuators
- [ ] Picks some data to see a graph of

Juerd, 2024-11-01:
This appears to be posted in the wrong repository, since these requirements seem very specific for a certain project.

mitra42, 2024-11-01:
Sorry - my mistake - had both repo's open ....

closed 2024-11-01 by mitra42