Juerd/ESP-WiFiSettings #43: Exposing the ssid

mitra42, 2024-11-18:
Would you have a problem exposing the ssid - you slurp it in portal and connect and spurt it on Save, I'm suggesting that instead of this being a local String in each of the functions, that its a global string exposed in the .h 

This would make the code in #42 simpler and also allow another tweak I'm making. 

I'm happy to do the Pull request if the idea is acceptable.

Juerd, 2024-11-18:
Sure.

In the code that builds the dropdown, the current SSID is in the variable `current` and it is compared to `ssid` which is each of the SSIDs. With this change, `current` would become `ssid` and what is now `ssid` could be renamed to `ssid_option' or something like that. Or maybe just `s`, the context of the code is clear enough.

Juerd, 2024-11-24:
Mitra Ardron skribis 2024-11-24 14:34 (-0800):
>@Juerd  - did you see I posted a PR for this.

Nope, I missed that.

Could you maybe also update the documentation? If not, I will do that myself.
-- 
Met vriendelijke groet, // Kind regards, // Korajn salutojn,

Juerd Waalboer  ***@***.***>
TNX

Juerd, 2024-11-25:
Fixed in #44.
Thanks!

closed 2024-11-25 by Juerd

mitra42, 2024-11-25:
I presume it takes a little while before the "official" library updates ?  I'm not sure how that process works (I submit libraries on npm, so presume its similar).

Juerd, 2024-11-25:
Honestly I don't know how often they pull changes.

mitra42, 2024-11-26:
Its there - working fine - thanks -